kas-lab/krr_bringup #11: Support mission configure file

Rezenders, 2025-02-25:
Objects
Mission type

chcorbato, 2025-03-19:
Following up on the meeting last week with @gavanderhoorn , we could implement our household robotic agent as task (I'll use task instead of mission) action server(s). Options I think of:
1. a single agent node that implements one action server per task. So depending on the task we want to request, a specific action client is used in our test script.
2. A single action server for all tasks, a field in the goal message identifies which task is it.

@gavanderhoorn @Rezenders @fghzamani @alexander-gabriel what do you think?

Recall: the possible tasks are:
- tidy up all objects by placing each one in the closest drop location
- tidy up all objests by placing them according to the apartment rules
- bring the an object to a specific drop location

gavanderhoorn, 2025-03-19:
The fact you 'have to' use three different bullets to describe what the tasks are to me implies already they are distinct actions.

Or at least two: *tidying up* and *fetching object*.

As I mentioned during our meeting, introspection is facilitated by using separate actions instead of using a single one / a few with (large amounts of) parameters. Parameters / action goal fields are invisible from the CLI. Actions name(spaces) are not.

chcorbato, 2025-03-19:
Thank you for the clear rationale for having multiple action servers.

The follow-up question I have is whether to implement those servers in separate nodes or the same one.
All those servers will access common information, both the world model (state of the world) and knowledge base (rules, types of things, and other general knowledge).

Are there any difficulties handling multiple action servers in a node?

gavanderhoorn, 2025-03-19:
> Are there any difficulties handling multiple action servers in a node?

No. Not that I'm aware of.

(assuming we're not talking about hundreds of action servers in a single node of course)

Rezenders, 2025-03-19:
> We need to provide them with templates. ROS Actions are not covered in previous courses.

Ok. Another point to discuss then: As a c++ ros node? as a python ros node?  as a c++ PlanSys node?

ArendJan, 2025-03-20:
> I would say as a Python node, because I think that is what the students learn in RSP.

No, this plan we switched back, we stayed with CPP last year

alexander-gabriel, 2025-03-20:
I'll implement the python action server template.

alexander-gabriel, 2025-03-20:
The generic version in Python now exists at https://github.com/kas-lab/krr_agent/tree/main/krr_agent_generic_task_server

I will extend it to task 3 next.

alexander-gabriel, 2025-03-20:
the task server is now extended to cover task 3, and further extended to include code that exemplifies calling an external reasoner on a dynamically created problem file in the /tmp directory.

Rezenders, 2025-03-20:
> bring an object to a specific drop location 
I am not sure how this task would work with the actions we planned. How would the robot know where to look for a specific object?

I believe this still needs to be addressed. With the current actions, this task cannot be achieved. The only possible way to do it, would be to try and pick the objects in an arbitrary order until the robot finds the correct object

> I would say as a Python node, because I think that is what the students learn in RSP.
However, it may be useful to give them a PlanSys based node, then the only option is C++
I propose to provide both templates:
@Rezenders the PlanSys-based node template and
maybe @alexander-gabriel can create a Python-based generic (not tied to a particular library to implement planning or reasoning) node.
What do you both think?

How would they use a Python node in their project with the technologies we presented during the course? 
If I am not wrong, we only showed how they can use PlanSys2 for task decision-making/planning with ROS, right?

Also, the PlanSys2 node example is not ready yet.

Rezenders, 2025-03-20:
> How would they use a Python node in their project with the technologies we presented during the course?

I see that you added an example on how to use leo3 here: https://github.com/kas-lab/krr_agent/blob/cdefa233007b33242a421313bb38c5d370c3b4ae/krr_agent_generic_task_server/krr_agent_generic_task_server/krr_agent_task_server.py#L158C4-L174C33

But I don't think it is very clear. How would they run what is outputted from leo3?

Rezenders, 2025-03-21:
> Using that, they can go through rooms in a sensible order, request the list of objects therein, and thus find the target, no?

They get the position of the objects, but no other info (such as object type etc).

chcorbato, 2025-03-21:
> > Using that, they can go through rooms in a sensible order, request the list of objects therein, and thus find the target, no?
> 
> They get the position of the objects, but no other info (such as object type etc).

Yes, the robot has a tedious task (what robots are for) of picking and checking if that is the object, then keep track of objects checked not to do a dumb search.  :-)

chcorbato, 2025-03-21:
@alex-quin-gabriel I want the option to include previously unknown objects (not in the table in the PDF) in a task. So that they can use of types/ontologies to reuse knowledge:

I added this in the PDF:
> information will be provided in the task message in the form of a tuple of strings, indicating the id of the object and the drop location where it normally belongs.

what do you think?

closed 2025-03-21 by chcorbato