kas-lab/krr_bringup #15: Add objects and placing positions to simulation (mvp)

Rezenders, 2025-02-25:
Predefined and not configurable
Object types: tv remote and tissue box

Rezenders, 2025-02-26:
> gazebo id

I suggest we add the objects with model names with the pattern `obj_{ID}_{TYPE}`. For example, `obj_1_spoon`, `obj_2_spoon`, `obj_3_remote` ...

And locations as `drop_{id}_{type}`. For example, `drop_1_bin`, `drop_2_box`.

chcorbato, 2025-02-26:
Good idea!
an alternative to "drop" could be "storage" or "place", so that they are not confused with actions.
But I see the point in the brevity of "drop", so any of those options are fine with me.

Rezenders, 2025-02-27:
Done at: https://github.com/kas-lab/robocup_home_simulation/pull/7

chcorbato, 2025-02-27:
@Rezenders here is my proposal for location and names of drop locations:
[image]

Rezenders, 2025-02-27:
Ok. I will update the simulation accordingly. 

What about the objects? So far, we have:

- 1 spoon in the kitchen
- 1 spoon and 1 bread in the bedroom

What about the drop location for each object?

- clean spoon in the tableware?
- dirty spoon in the dishwasher?
- bread? ( I can remove the bread and put something else)

We could get models from https://app.gazebosim.org/. I think most of the models have a [Creative Commons Attribution 4.0 International](http://creativecommons.org/licenses/by/4.0/) license. Such as:

- https://app.gazebosim.org/search;q=razer
- https://app.gazebosim.org/GoogleResearch/fuel/models/Eat_to_Live_The_Amazing_NutrientRich_Program_for_Fast_and_Sustained_Weight_Loss_Revised_Edition_Book
- https://app.gazebosim.org/GoogleResearch

chcorbato, 2025-02-27:
I propose to drop the bread and other trash in a new `drop_trashcan`

[image]

alexander-gabriel, 2025-02-27:
can we get the tvremote back?
ideally placed anywhere but the livingroom.

Rezenders, 2025-02-27:
> I propose to drop the bread and other trash in a new drop_trashcan

Done: [15d5787](https://github.com/kas-lab/robocup_home_simulation/commit/15d5787e89c5560a4acd2dcf419a8d7075fa235b)

Rezenders, 2025-02-27:
> can we get the tvremote back?
> ideally placed anywhere but the livingroom.

If we find a remote model somewhere

closed 2025-02-27 by Rezenders

chcorbato, 2025-02-28:
Shall we replace the tv remote with a different object? Anything available that makes sense in a living room and the size is reasonable for mirte to pick @Rezenders @alexander-gabriel