kas-lab/krr_bringup #16: Grasp plugin not loaded until closing gazebo

ArendJan, 2025-02-25:
I ran: `ros2 launch mirte_gazebo gazebo_mirte_master_navigation.launch.py 
`

Log:

```

[planner_server-6] [INFO] [1740515617.003003030] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-12] [INFO] [1740515617.029759155] [rviz]: Message Filter dropping message: frame 'lidar_link' at time 5586.081 for reason 'discarding message because the queue is full'
[rviz2-12] [INFO] [1740515617.094639591] [rviz]: Message Filter dropping message: frame 'odom' at time 5585.147 for reason 'discarding message because the queue is full'
[rviz2-12] [INFO] [1740515617.094861751] [rviz]: Message Filter dropping message: frame 'odom' at time 1740515611.082 for reason 'discarding message because the queue is full'
[rviz2-12] [INFO] [1740515617.126356850] [rviz]: Message Filter dropping message: frame 'lidar_link' at time 5586.148 for reason 'discarding message because the queue is full'
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rviz2-12] [INFO] [1740515617.188720328] [rclcpp]: signal_handler(signum=2)
[twist_mux-20] [INFO] [1740515617.188721043] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-16] [INFO] [1740515617.188722267] [rclcpp]: signal_handler(signum=2)
[controller_server-4] [INFO] [1740515617.188765396] [rclcpp]: signal_handler(signum=2)
[amcl-2] [INFO] [1740515617.188824576] [rclcpp]: signal_handler(signum=2)
[amcl-2] [INFO] [1740515617.188955144] [amcl]: Running Nav2 LifecycleNode rcl preshutdown (amcl)
[amcl-2] [INFO] [1740515617.189053688] [amcl]: Deactivating
[amcl-2] [INFO] [1740515617.189080488] [amcl]: Destroying bond (amcl) to lifecycle manager.
[behavior_server-7] [INFO] [1740515617.188835902] [rclcpp]: signal_handler(signum=2)
[lifecycle_manager-11] [INFO] [1740515617.188936932] [rclcpp]: signal_handler(signum=2)
[lifecycle_manager-11] [INFO] [1740515617.190564334] [lifecycle_manager_navigation]: Running Nav2 LifecycleManager rcl preshutdown (lifecycle_manager_navigation)
[planner_server-6] [INFO] [1740515617.188997080] [rclcpp]: signal_handler(signum=2)
[planner_server-6] [INFO] [1740515617.189301039] [planner_server]: Running Nav2 LifecycleNode rcl preshutdown (planner_server)
[planner_server-6] [INFO] [1740515617.189382988] [planner_server]: Destroying bond (planner_server) to lifecycle manager.
[planner_server-6] [INFO] [1740515617.189645221] [global_costmap.global_costmap]: start
[planner_server-6] [ERROR] [1740515617.190109926] [global_costmap.global_costmap]: Caught exception in callback for transition 13
[planner_server-6] [ERROR] [1740515617.190123570] [global_costmap.global_costmap]: Original error: could not create subscription: rcl node's context is invalid, at ./src/rcl/node.c:428
[planner_server-6] [ERROR] [1740515617.190146281] [global_costmap.global_costmap]: Failed to finish transition 3. Current state is now: errorprocessing (Could not publish transition: publisher's context is invalid, at ./src/rcl/publisher.c:389, at ./src/rcl_lifecycle.c:368)
[planner_server-6] [INFO] [1740515617.190157659] [planner_server]: Activating plugin GridBased of type NavfnPlanner
[planner_server-6] [ERROR] [1740515617.190230622] [planner_server]: Caught exception in callback for transition 13
[planner_server-6] [ERROR] [1740515617.190236185] [planner_server]: Original error: could not create service: rcl node's context is invalid, at ./src/rcl/node.c:428
[planner_server-6] [ERROR] [1740515617.190246134] [planner_server]: Failed to finish transition 3. Current state is now: errorprocessing (Could not publish transition: publisher's context is invalid, at ./src/rcl/publisher.c:389, at ./src/rcl_lifecycle.c:368)
[rviz2-12] 
[rviz2-12] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[rviz2-12] This error state is being overwritten:
[rviz2-12] 
[rviz2-12]   'rcl node's context is invalid, at ./src/rcl/node.c:428'
[rviz2-12] 
[rviz2-12] with this new error message:
[rviz2-12] 
[rviz2-12]   'the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at ./src/rcl/wait.c:130'
[rviz2-12] 
[rviz2-12] rcutils_reset_error() should be called after error handling to avoid this.
[rviz2-12] <<<
[rviz2-12] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[rviz2-12]   what():  failed to initialize wait set: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at ./src/rcl/wait.c:130
[controller_server-4] [INFO] [1740515617.189920627] [controller_server]: Running Nav2 LifecycleNode rcl preshutdown (controller_server)
[controller_server-4] [INFO] [1740515617.190022516] [controller_server]: Deactivating
[controller_server-4] [INFO] [1740515617.190056822] [MPPIController]: Deactivated MPPI Controller: FollowPath
[controller_server-4] [INFO] [1740515617.190079381] [local_costmap.local_costmap]: Deactivating
[waypoint_follower-9] [INFO] [1740515617.188826714] [rclcpp]: signal_handler(signum=2)
[waypoint_follower-9] [INFO] [1740515617.191086112] [waypoint_follower]: Running Nav2 LifecycleNode rcl preshutdown (waypoint_follower)
[waypoint_follower-9] [INFO] [1740515617.191178778] [waypoint_follower]: Cleaning up
[lifecycle_manager-3] [INFO] [1740515617.189499926] [rclcpp]: signal_handler(signum=2)
[lifecycle_manager-3] [INFO] [1740515617.189607249] [lifecycle_manager_localization]: Running Nav2 LifecycleManager rcl preshutdown (lifecycle_manager_localization)
[lifecycle_manager-3] [INFO] [1740515617.189656048] [lifecycle_manager_localization]: Terminating bond timer...
[bt_navigator-8] [INFO] [1740515617.189752001] [rclcpp]: signal_handler(signum=2)
[bt_navigator-8] [INFO] [1740515617.191890549] [bt_navigator]: Running Nav2 LifecycleNode rcl preshutdown (bt_navigator)
[bt_navigator-8] [INFO] [1740515617.192011570] [bt_navigator]: Cleaning up
[smoother_server-5] [INFO] [1740515617.189822686] [rclcpp]: signal_handler(signum=2)
[smoother_server-5] [INFO] [1740515617.190204676] [smoother_server]: Running Nav2 LifecycleNode rcl preshutdown (smoother_server)
[smoother_server-5] [INFO] [1740515617.190300346] [smoother_server]: Deactivating
[smoother_server-5] [INFO] [1740515617.190329915] [smoother_server]: Destroying bond (smoother_server) to lifecycle manager.
[smoother_server-5] [INFO] [1740515617.200593417] [smoother_server]: Cleaning up
[velocity_smoother-10] [INFO] [1740515617.190313840] [rclcpp]: signal_handler(signum=2)
[velocity_smoother-10] [INFO] [1740515617.190758469] [velocity_smoother]: Running Nav2 LifecycleNode rcl preshutdown (velocity_smoother)
[velocity_smoother-10] [INFO] [1740515617.190907662] [velocity_smoother]: Cleaning up
[... 49 lines cut ...]
[controller_server-4] [INFO] [1740515617.396545651] [controller_server]: Destroying
[INFO] [twist_mux-20]: process has finished cleanly [pid 63976]
[ERROR] [ros2-19]: process has died [pid 63972, exit code 2, cmd 'ros2 topic pub /zero_cmd_vel geometry_msgs/msg/Twist {} -r 100'].
[INFO] [velocity_smoother-10]: process has finished cleanly [pid 63885]
[INFO] [map_server-1]: process has finished cleanly [pid 63867]
[INFO] [smoother_server-5]: process has finished cleanly [pid 63875]
[INFO] [waypoint_follower-9]: process has finished cleanly [pid 63883]
[INFO] [amcl-2]: process has finished cleanly [pid 63869]
[INFO] [lifecycle_manager-3]: process has finished cleanly [pid 63871]
[ERROR] [rviz2-12]: process has died [pid 63897, exit code -6, cmd 'rviz2 -d /opt/ros/humble/share/nav2_bringup/rviz/nav2_default_view.rviz'].
[INFO] [bt_navigator-8]: process has finished cleanly [pid 63881]
[INFO] [behavior_server-7]: process has finished cleanly [pid 63879]
[ERROR] [planner_server-6]: process has died [pid 63877, exit code -6, cmd '/opt/ros/humble/lib/nav2_planner/planner_server --ros-args --log-level info --ros-args -r __node:=planner_server --params-file /tmp/launch_params_j05zcptv -r /tf:=tf -r /tf_static:=tf_static'].
[INFO] [controller_server-4]: process has finished cleanly [pid 63873]
[INFO] [gzclient-14]: process has finished cleanly [pid 63902]
[INFO] [gzserver-13]: process has finished cleanly [pid 63900]
[gzserver-13] 
[gzserver-13] 
[lifecycle_manager-11] [ERROR] [1740515619.196698107] [lifecycle_manager_navigation]: Failed to change state for node: planner_server. Exception: planner_server/get_state service client: async_send_request failed.
[lifecycle_manager-11] [ERROR] [1740515619.196768997] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
[lifecycle_manager-11] [INFO] [1740515619.200084881] [lifecycle_manager_navigation]: Destroying lifecycle_manager_navigation
[INFO] [lifecycle_manager-11]: process has finished cleanly [pid 63888]
➜  krr_base_ws 

```

chcorbato, 2025-02-26:
@ArendJan can you please discuss this with @fghzamani to decide what and how needs to be fixed?

Rezenders, 2025-02-26:
I will work on this now

Rezenders, 2025-02-26:
Hi @ArendJan,

I think the plugin is being loaded properly even though the messages are published only when gazebo shuts down.

```
Singularity:krr_ws> gz topic -l | grep contacts
/gazebo/default/mirte/contacts
/gazebo/default/physics/contacts
Singularity:krr_ws> gz topic -i /gazebo/default/mirte/contacts
Type: gazebo.msgs.Contacts

Publishers:
	145.94.204.79:40151

Subscribers:
	145.94.204.79:40151

Singularity:krr_ws> 
```

List of plugins in gazebo client:

[image]

However, I have no idea how to test whether it actually works. I was trying to figure it out but I couldn't.

Rezenders, 2025-02-26:
I will mark this as closed for now. If there are issues with the plugin itself lets open a new issue

closed 2025-02-26 by Rezenders

ArendJan, 2025-02-26:
@Rezenders we also don't need it for krr

chcorbato, 2025-02-26:
It will be useful in a future more realistic simulation of the mirte master grasping, thank you @Rezenders !