kas-lab/krr_bringup #2: Pick action

Rezenders, 2025-02-25:
Create a Pick action server.
Action parameters: object id
Output: Sucess or Failure

Return failure if the robot is already holding an object or if the robot is further than `dist` from the object

chcorbato, 2025-02-26:
Existing Gazebo service useful to get object information:
https://docs.ros.org/en/humble/p/gazebo_msgs/interfaces/srv/GetEntityState.html

Elvin is creating code to get all objects(aka models) here #4

Rezenders, 2025-02-26:
@ArendJan how do I know which gazebo model was picked-up? Are you going to do something similar to GraspFix plugin and attach the model to the `arm_Wrist` link?
For example, `obj_1` was picked

ArendJan, 2025-02-26:
Dunno yet, can add that!

ArendJan, 2025-02-26:
Second part: I'll probably use the boeing plugin for attaching

Rezenders, 2025-02-26:
This one? https://github.com/Boeing/gazebo_model_attachment_plugin
For now I will assume you will use this plugin and set the [`joint_name`](https://github.com/Boeing/gazebo_model_attachment_plugin/blob/9f9278284ba91f3632caaf5742c479fdf4e63182/README.md?plain=1#L78)  to `attachament_joint`.
Then, if you do something different, let me know. I will use this assumption to work on #5

Rezenders, 2025-02-27:
Just FYI, I merged what @fghzamani has done until now, but it is not working 100% of the time.
For some reason, sometimes it works, and sometimes it doesn't.
We were trying to figure out the reason but didn't find out. 
I suspect the problem might be in the plugin (trying to take the blame away from us hehe)
I have no more brain energy, so I will stop helping forough for now. I will just finish pushing everything that was done, so we can rebuild the singularity image.

Rezenders, 2025-02-27:
> For some reason, sometimes it works, and sometimes it doesn't.
> We were trying to figure out the reason but didn't find out.
> I suspect the problem might be in the plugin (trying to take the blame away from us hehe)

It was indeed the plugin. @ArendJan found and fixed the bug in the PR https://github.com/kas-lab/krr_bringup/pull/20

I will leave this issue open, as the action server is taking the `object_id` as input. This was due to a misalignment between us. See the **Edit** message in this issue description

chcorbato, 2025-02-28:
Amazing work team. it is acceptable that things work between 50%-100% of the time.
As long as the students can do something with it, we can involve them in debugging it with us.
Great work team!!

Rezenders, 2025-02-28:
> I will leave this issue open, as the action server is taking the object_id as input. This was due to a misalignment between us. See the Edit message in this issue description

I will leave this open for now because of the comment above

chcorbato, 2025-03-14:
@fghzamani this issue is open, see previous comment from @Rezenders 
This needs to be solved with urgently, next week we need to get to a stable useable course project set up and detailed description.
Please comment if there are issues blocking this.

fghzamani, 2025-03-17:
I just got back on this issue. But I don't remember the agreement we made about it if the pick server does not need the `obj_id` as an input, then what would be its input? I read the #5 but did not understand how to update the input of the server. I need more clarification here. Can someone explain? @chcorbato @Rezenders

Rezenders, 2025-03-18:
@fghzamani, the pick server should have no input. It should simply pick the object in front of the robot if its distance is lower than a predefined threshold.

This is required since the service `get objects in the room` (https://github.com/kas-lab/krr_bringup/issues/4) returns the position of the objects and not their ids.

chcorbato, 2025-03-19:
> [@chcorbato](https://github.com/chcorbato) should we make this backward compatible?
> Maybe creating a different service for the new version of the pick action? Another option is to leave the `object_id` as an optional argument, when it is defined the robot tries to pick that specific object, when it is not defined it tries to pick the object in its front

I agree with making the `object_id` optional. Does anyone sees any major issue with that @fghzamani @alexander-gabriel ?

alex-quin-gabriel, 2025-03-20:
sounds like a good compromise to me. also somewhat makes sense.

Rezenders, 2025-03-20:
Done : https://github.com/kas-lab/krr_mirte_skills/pull/6