kas-lab/krr_bringup #4: Get objects in the room action

Rezenders, 2025-02-25:


chcorbato, 2025-02-26:
I propose we call all objects "entities".
Gazebo calls them sometime smodels, but that will be confusing in KRR course.

chcorbato, 2025-02-26:
About this :
> Probably we should store somewhere a database of task entities for all servers to use consistently.

@EGAlberts @Rezenders @fghzamani @alexander-gabriel what do you think?

Rezenders, 2025-02-26:
@EGAlberts  @chcorbato  this was assigned to me before. I was going to start working on it now.
@EGAlberts will you take care of this then?

Rezenders, 2025-02-26:
> Probably we should store somewhere a database of task entities for all servers to use consistently.

I don't understand what you mean by this

chcorbato, 2025-02-26:
> [@EGAlberts](https://github.com/EGAlberts) [@chcorbato](https://github.com/chcorbato) this was assigned to me before. I was going to start working on it now. [@EGAlberts](https://github.com/EGAlberts) will you take care of this then?

Sorry, I forgot and did not see it here on the issue, so gave it to Elvin. He is doing it.

chcorbato, 2025-02-26:
> > Probably we should store somewhere a database of task entities for all servers to use consistently.
> 
> I don't understand what you mean by this

We need to store somewhere state information about the objects that we are defining for the course project task. E.g. the spoon may be dirty, and then it needs to go somewhere different than if it were clean.
There presents two related requirements:
- A server needs to provide information about the spoon being clean or not.
- This information needs to be stored somewhere.

Discussing it with Elvin, there is only minimal "semantic" information about objects that we need to provide: basically the status in terms of cleanliness and the position.
The position information is provided directly by the gazebo service.
So to add the cleanliness information, rather than creating some database, we could use the gazebo model itself by using the model name data and appending that state information to it: e.g. `spoon1_dirty`
We then put the responsilibity on the students to parse this information into the KB of their KRR solution.

FYI @EGAlberts @alexander-gabriel @fghzamani

Rezenders, 2025-02-26:
Living room:

- -3.158871  -4.849809
- -9.560931 -4.792851
- -9.365241 1.888928
- -3.252633  1.818826

Rezenders, 2025-02-26:
Office:

- -3.252632 2.044113
- -9.363742 2.110951
- -9.353752 7.379319
- -3.243108  7.376032

Rezenders, 2025-02-26:
Bedroom:
 
- 2.766750 2.188229
- -3.026235 2.058432
- -2.993710 7.358111
- 2.769169  7.380637

EGAlberts, 2025-02-26:
Indeed, they are four corners of each room which will be added to a dictionary in the service. The node right now has the missing last room coded into it already.

Rezenders, 2025-02-27:
solved in PR https://github.com/kas-lab/krr_mirte_skills/pull/1

closed 2025-02-27 by Rezenders