kas-lab/krr_bringup #7: Add drop locations to simulatio

Rezenders, 2025-02-25:


chcorbato, 2025-02-26:
I am deleting this issue, since we have one with all elements to be added to the simulation

fghzamani, 2025-03-17:
Can you provide more information about what is exactly needed regarding the configurable storage locations? @chcorbato

Rezenders, 2025-03-17:
If I remember correctly, the idea was that we could provide a configuration file to the simulation indicating which drop locations should be present and their locations.  For example, `drop_trash` at `{x: 1.0, y: 2.0}` etc.

I am not sure this is still required. Are we gonna have fixed drop locations as we already have, or do we still need configurable drop locations? What do you think @chcorbato @alexander-gabriel @fghzamani @buraksisman ?

chcorbato, 2025-03-18:
@alexander-gabriel do you have an idea on how to make them configurable?
alternatively, we manually create a few worlds with different drop locations, we give a subset to the students to test their solution, and reserve a few for us to evaluate their solutions.

chcorbato, 2025-03-19:
> > do you have an idea on how to make them configurable?
> 
> * [Procedural Generation for Gazebo](https://boschresearch.github.io/pcg_gazebo) (ROS 1 though) or
> * [Gazebo ERB Templates](https://gazebosim.org/api/sim/9/erbtemplate.html)
> 
> perhaps?
> 
> Might be too much work for now though.

Sounds like better to manually create a few test worlds, i.e. 4 for students to test, 3 or so for evaluation.

Rezenders, 2025-03-19:
As we agreed today, this is not needed anymore. I am closing this issue.

chcorbato, 2025-03-19:
Reopen to decide whether to address it in the future, not to forget the idea.