kas-lab/krr_bringup #21: remove mirte parts from laser/lidar readings

chcorbato, 2025-02-28:
Mentioned also [here](https://github.com/kas-lab/mirte-ros-packages/pull/2#issuecomment-2688415949)
This probably should be addressed setting up a [laser_filter](https://github.com/ros-perception/laser_filters), see [answer by Mazensky](https://robotics.stackexchange.com/questions/93739/how-to-eliminate-robots-parts-from-lidar-data/93740#93740)

Probably this should be part of the Mirte software setup, at least mirte-gazebo, maybe even also for the real robot, just different configuration. In that case, please transfer the issue to the corresponding repo @ArendJan

chcorbato, 2025-03-16:
So far this does not seem critical to have. But probably the more we use the simulation the more it is preferred to have Nav2 configured as good as possible

ArendJan, 2025-03-17:
I've just added laserscan filter based on distance. https://github.com/kas-lab/mirte-gazebo/pull/2

fghzamani, 2025-03-17:
You are right. This is a relative parameter for this matter, although it is "the maximum distance to consider skipping for (m)" based on the Nav2 documentation: https://docs.nav2.org/configuration/packages/configuring-amcl.html 
@Rezenders

Rezenders, 2025-03-18:
Ok, I will check [PR #2](https://github.com/kas-lab/mirte-gazebo/pull/2) first, then close this issue

Rezenders, 2025-03-18:
Done: https://github.com/kas-lab/mirte-gazebo/pull/2 and https://github.com/kas-lab/mirte-ros-packages/pull/4
Thanks @ArendJan

closed 2025-03-18 by Rezenders