kas-lab/krr_bringup #26: Update Course project description

chcorbato, 2025-03-05:
We need to polish the project description with all the details and final specs

chcorbato, 2025-03-27:
I added instruction for the technical report

chcorbato, 2025-03-27:
@buraksisman you are presenting this tomorrow, if anything is missing or needs clarification please tag here whoever you think can help with that.

alex-quin-gabriel, 2025-03-27:
I've updated the run instructions for task 3 to point to the correct test scenarios.

chcorbato, 2025-03-31:
I also made updates on Friday.

@buraksisman ca you please update issue status? Can we close it?

buraksisman, 2025-04-01:
I think the only issue that left is updating Table 1. It still has text in red color. Moreover, there are some definitions that I could not see in the simulation. For instance, general drop point. So that, I think it is still not completed. @fghzamani can you update it or you can pass me relevant information and than I can complete as well.