kas-lab/krr_bringup #31: start the simulation providing a 2D pose estimate for Nav2/amcl

chcorbato, 2025-03-17:
Start the simualtion of mirte initializing the 2D pose estimate for AMCL, so that it is not necessary to do it manually on Rviz

chcorbato, 2025-03-17:
I have tried to do it myself without success.

adding the initialization in the params for amcl did tno work for me. That would be (according to [nav2 doc](https://docs.nav2.org/configuration/packages/configuring-amcl.html#parameters)) in 
`mirte_navigation/params/mirte_nav2_params.yaml`

```
amcl:
  ros__parameters:
     set_initial_pose: true
     initial_pose:
        x: 0.0
        y: 0.0
        z: 0.0
        yaw: 0.0
```

I tried to add a static publisher, and the only thing that did something (suggested by Copilot) was to to add an `initial_pose_publisher` in the launch file:

```
    # Static transform publisher for initial pose
    initial_pose_publisher = Node(
        package='tf2_ros',
        executable='static_transform_publisher',
        name='initial_pose_publisher',
        output='screen',
        arguments=['0.0', '0.0', '0.0', '0.0', '0.0', '1.0', 'map', 'base_link']
    )

    return LaunchDescription([
        ...
        initial_pose_publisher
    ])
```

This made the error about the map frame not existing dissapear in RViz, but did not initialized AMCL.
I guess if we change that publisher from a tf2 into a Pose publisher maybe we get it.

fghzamani, 2025-03-17:
I fixed this problem and sent a pull request to the mirte_navigation repo. Could you check and see if it is okay to accept my pull request? @Rezenders

Rezenders, 2025-03-17:
@fghzamani There are no open pull requests in the mirte_navigation repo about this: https://github.com/kas-lab/mirte_navigation/pulls?q=is%3Apr+

Perhaps it is in a different repo, can you let me know? I tried finding it, but I didn't find it.

Rezenders, 2025-03-17:
Done: https://github.com/kas-lab/mirte_navigation/pull/2

closed 2025-03-17 by Rezenders

alex-quin-gabriel, 2026-03-24:
this is fixed in https://github.com/kas-lab/mirte_navigation/pull/17