kas-lab/krr_bringup #32: Decisions on project tasks and rules

chcorbato, 2025-03-18:
- Issues with current rules:
  - [ ] an object next to the doorway in an adjacent room would be unknown to the robot, with the risk of running over it.
  - [ ] objects in the same room as the robot may not be detected by the sensors for navigation, so the robot may move other them

fghzamani, 2025-03-26:
I can not find the table of the objects and their type or expected drop location. Do we have such a table? @chcorbato @Rezenders @alex-quin-gabriel @buraksisman

chcorbato, 2025-03-26:
Table 1 in the overleaf you have open 😜

chcorbato, 2025-03-27:
Some decisions this week have been discussed on Teams.
@buraksisman are you or @Rezenders adding the corresponding clarifications on the PDF?
Once that is done, I think we can close this issue.

Rezenders, 2025-03-27:
I updated the pdf with the information I remembered/assigned to me. I have nothing else to update.
We still need to update the object types and drop location info

closed 2025-03-31 by chcorbato