kas-lab/krr_bringup #33: an object next to the doorway in an adjacent room would be unknown to the robot, with the risk of running over it.

chcorbato, 2025-03-19:


alexander-gabriel, 2025-03-20:
We could solve this by introducing 'doorway areas' as quasi rooms, i.e. rectangles as wide as the doorway extending a meter or so in the rooms on each side...
the service that returns the objects in a room could then return the objects in a given doorway area.

chcorbato, 2025-03-21:
to discuss at 14:30

We could reuse `GetObjectsInRoom.srv` adding a parameter to the request for the name of the doorway:
-  if not present, the service works as before returning the poses of objects in the room.
- if present, then
   - if the robot is in a room adjacent to the doorway, it returns the poses of objects in the doorway
   - if the robot is NOT in a room adjacent to the doorway, it returns `failure`

Rezenders, 2025-03-21:
https://github.com/kas-lab/krr_bringup/issues/60#issuecomment-2743483639

EGAlberts, 2025-03-21:
I made this happen as follows: The getobjects in room now has two returns. the obj list for what's in the room, and a list of DoorwayObjects, which are a tuple of names of doorways and objects in those doorways. See my PR: https://github.com/kas-lab/krr_mirte_skills/pull/7

closed 2025-03-21 by Rezenders