kas-lab/krr_bringup #34: objects in the same room as the robot may not be detected by the sensors for navigation, so the robot may move other them

chcorbato, 2025-03-19:


Rezenders, 2025-03-21:
We agreed that we will allow this to happen. We will disable the collision of the objects that have to be picked.

chcorbato, 2025-03-21:
This needs to be written in the overleaf

Rezenders, 2025-03-24:
Hmm, actually, disabling the collision of objects is not that simple as they will "fall" from the ground

chcorbato, 2025-03-24:
@EGAlberts @fghzamani any of you available to help with this?

Rezenders, 2025-03-24:
We can use bitmasks: https://classic.gazebosim.org/tutorials?tut=collide_bitmask&cat=physics

Rezenders, 2025-03-24:
To use collision bitmask, we need to define a collision bitmask for mirte (for all links? or can we do it for the model?) and a different collision_bitmask for the objects 

E.g.:
Mirte:
```
<surface>
        <contact>
          <collide_bitmask>0x01</collide_bitmask>
        </contact>
      </surface>
```

Objects:
```
<surface>
        <contact>
          <collide_bitmask>0x02</collide_bitmask>
        </contact>
      </surface>
```

Rezenders, 2025-03-24:
> Can you disable their gravity?

Hmm, that is true. I could try that.

Rezenders, 2025-03-24:
Gravity off + collision off works. But then we need to position the objects very well:

[image]

chcorbato, 2025-03-24:
You can move them manually with the gui and save the world, no need to guess poses

Rezenders, 2025-03-24:
> Gravity off + collision off works

Actually, it doesn't work. Because when we perform the place action, we need the object to be subject to gravity so it can fall to the ground

chcorbato, 2025-03-24:
So let the robot collide with objects and hope that it does not mess up with the simulation (e.g. object entangled in Mirte).
No penalties for driving over objects.
Agree @Rezenders ?

Rezenders, 2025-03-24:
> o let the robot collide with objects and hope that it does not mess up with the simulation (e.g. object entangled in Mirte).
No penalties for driving over objects.
Agree @Rezenders ?

OK.

I tried adding collide_bitmasks to mirte and the objects. It seems that it worked for the wheels, but there is something in the body still colliding. Only hits the hat and bread now.

https://github.com/kas-lab/mirte-ros-packages/tree/collide_bitmask
https://github.com/kas-lab/krr_agent/tree/collide_bitmask

I didn't merge it because it doesn't fully work

EGAlberts, 2025-03-24:
Increasingly hacky suggestion: modify the gravity being in effect at runtime, manipulating it depending on whether you're meant to drop it of not

Rezenders, 2025-03-25:
@ArendJan suggested deactivating Mirte's collision. Since it is using the Planar plugin it doesn't need to be in contact with the ground. This seems like the easiest solution.
The downside is that mirte will no collide with anything else, but assuming nav2 is correctly setup, it won't collide anyway

EGAlberts, 2025-03-25:
It is probably for the best, it prevents any breakdancing if nav2 does fail

Rezenders, 2025-03-27:
I deactivated mirte collisions, seems to work. Can you please review https://github.com/kas-lab/mirte-ros-packages/pull/5 @ArendJan?

closed 2025-03-27 by Rezenders