kas-lab/krr_bringup #37: [Task 1] Create world for Scenario 3

Rezenders, 2025-03-19:
Scenario 3: There is a new drop location in one of the rooms that is closer to an object than previous existing drop locations.

Rezenders, 2025-03-19:
Done: https://github.com/kas-lab/krr_agent/commit/c0566a09e2e36e6cb815022733819b1762b51ce2

I named the new drop location as `drop_8_hats`

[image]

To test this world, you need to clone the [krr_agent](https://github.com/kas-lab/krr_agent) repo into the krr workspace, rebuild the workspace, and run:

```bash
ros2 launch krr_agent task1.launch.py world:=scenario3.world
```

chcorbato, 2025-03-19:
> To test this world, you need to clone the [krr_agent](https://github.com/kas-lab/krr_agent) repo into the krr workspace, rebuild the workspace, and run:
> 
> ros2 launch krr_agent task1.launch.py world:=scenario3.world

I follow these instructions and the ws builds without issues, but I do not see the new objects and drop locations in Gazebo (also not in the list of models in Gazebo)

Rezenders, 2025-03-20:
The testing instructions were incomplete.
To test it, you need to download the newest singularity image: https://filesender.surf.nl/?s=download&token=d44d3b76-e5e7-40ee-9a6c-b96b16ed4bf7

From the singularity image, update your workspace:
```
cd ~/krr_ws
vcs pull src
```

Clone krr_agent
```
cd ~/krr_ws/src
git clone https://github.com/kas-lab/krr_agent.git
```

Build it again:
```
cd ~/krr_ws
colcon build --symlink-install
source install setup.bash
```

Then run it:
```
ros2 launch krr_agent task1.launch.py world:=scenario1.world
ros2 launch krr_agent task1.launch.py world:=scenario2.world
ros2 launch krr_agent task1.launch.py world:=scenario3.world
```

buraksisman, 2025-03-20:
this line should be:

source install/setup.bash

chcorbato, 2025-03-20:
After locally fixing the issues, the gazebo no longer launches for me; I continuously get the `gazebo is not responding` error. Normally, I get it once, then I click wait and it loads. Now I stopped trying after waiting 5 times:

[image]

@Rezenders @ArendJan help needed.
@buraksisman, @fghzamani  could you please check if this also happens to you or it is a very particular performance issue of my hw setup?

chcorbato, 2025-03-20:
> After locally fixing the issues, the gazebo no longer launches for me; I continuously get the `gazebo is not responding` error. Normally, I get it once, then I click wait and it loads. Now I stopped trying after waiting 5 times:

Ok, so it is a performance issue, it loads apparently correctly after waiting some time, so I will close this issue and open one about the performance to address in the future.

closed 2025-03-20 by chcorbato