kas-lab/krr_bringup #39: [Task 1] Add the size of the drop locations to the DropLocation message

Rezenders, 2025-03-19:
Add the size of the drop locations to the [DropLocation](https://github.com/kas-lab/krr_mirte_skills/blob/main/krr_mirte_skills_msgs/msg/DropLocation.msg) message. 
We could either add the size of the sides (assume it is a rectangle), or add the pose of each corner.

chcorbato, 2025-03-19:
@fghzamani can you help me with this?
> We need to check which part of the drop location the gazebo pose corresponds to. The center? One of the corners?

and also with the dimensions of it? Then I will update the overleaf.

chcorbato, 2025-03-21:
kind reminder @fghzamani, can you please provide me with the pose and dimensions of the drop locations? Also, what is the relative position of their coordinate frame? The center of the drop location? one of the corners? which one?

fghzamani, 2025-03-21:
Sure. The size of all the drop locations is a square of side length 1m. It has a height of 0.01 m.  
So the vector of its size is (1,1,0.01).
Regarding the pose of the drop location, they are listed below:
 
- drop_1_dishwasher: <pose>1.5 0 0 0 0 0</pose>
- drop_2_tableware: <pose>0.356888 -1.47612 0 0 -0 0</pose>
- drop_3_livingroom: <pose>-4 -3 0 0 0 0</pose>
- drop_4_toys: <pose>-8 3 0 0 0 0</pose>
- drop_5_laundry: <pose>-2.5096 6.78965 0 0 -0 0</pose>
- drop_6_bedroom: <pose>2.15269 6.51919 0 0 -0 0</pose>
- drop_7_trash: <pose>-8.64577 1.30639 0 0 -0 0</pose>
Regarding your last question about the relative position of their coordinate frame, I have to say the pose specified for a model is applied to the model's geometric center. For drop locations squares (size 1m x 1m x 0.01m):
 
    The coordinate origin is exactly at the centre of the square geometry.
    Thus, from this pose, the square extends:
        0.5m in the positive and negative X directions.
        0.5m in the positive and negative Y directions.
        Vertically, it extends 0.005m upwards (Z-axis positive) and 0.005m downwards (Z-axis negative) from Z=0.

chcorbato, 2025-03-21:
Done in appendix Drop Locations https://www.overleaf.com/project/6707e71df8b0c1f2941f03c6

closed 2025-03-21 by chcorbato