kas-lab/krr_bringup #42: document distance to pick parameter

Rezenders, 2025-03-19:


chcorbato, 2025-03-19:
@fghzamani can you give me this info, or point me to the code where I can find it?

fghzamani, 2025-03-24:
Regarding the pick distance constraints:

-     As mentioned by Arend, the pick service accepts a maximum distance of 0.5 m if the object is positioned in front of the robot.

-     However, if the object is positioned to the side or behind the robot, the allowed pick distance is reduced to 0.1 m.

The following image clarifies this behavior:

[image]

Example from the image:

In this scenario, three objects are present. The blue spoon is the closest object, but it's located aside the robot, approximately 0.28 m away. According to the implemented logic of the pick service, this position is classified as not in front (i.e., behind or to the side). Consequently, because the spoon is farther than 0.1 m, the pick service would not allow the robot to pick it.
Is this OK? @chcorbato

fghzamani, 2025-03-24:
This explanation reminds me of the tuning of the goal tolerance parameters. If we increase the yaw_tolarance to 3.14 rad, then there might be situations where the distance between the robot and the object is 0.4 which is in the allowable range of pick service, but since the orientation is not restricted, let's say the robot is in an orientation where the object is aside.  This means the robot can not pick it. There are three ways to resolve this inconsistency:

1. The students calculate the orientation needed for a successful pick using this formula:
`Robot pose: (xr,yr,θr), Object pose: (xo,yo)
and θgoal​=arctan2(yo​−yr​,xo​−xr​)
`
And then with this, we yaw_goal_tolarance could be less like 0.5 since the right orientation(upfront) is being calculated. 

2.  Changing the pick place service to consider 0.5 distance also from aside of the robot as well. And then there is no need to change the yaw_goal_tolerance(it can remain 3.14).
These are my brainstorming. What do you think? @chcorbato @Rezenders

chcorbato, 2025-03-24:
BTW I think I already documented this in the PDF, maybe forgot to close the issue.
@buraksisman can you confirm? Please reopen the issue only if needed.