kas-lab/krr_bringup #43: Create a solution for transforming an object pose to nav2 pose

Rezenders, 2025-03-19:
Create a solution for transforming an object pose that is received by calling the `get_objects_in_room` service, to a Mirte Pose to be sent to nav2 as a goal pose to pick the object. That is, the Mirte goal pose should be slightly offset from the object and contain the proper orientation

chcorbato, 2025-03-19:
We solve this with a service `compute_reach_pose` that takes as input argument the object pose and the mirte pose.
Could you help with this next week Monday @EGAlberts ?

EGAlberts, 2025-03-21:
@chcorbato shall I still do this? I was you moved it out of the Todo.

chcorbato, 2025-03-21:
Not sure this is still relevant @Rezenders ?

Maybe better if you could help create more test gazebo worlds, since that is still not ready afaik

Rezenders, 2025-03-24:
This is not strictly necessary since we will disable the objects' collision, but it would be nice to have

fghzamani, 2025-03-24:
I think this is highly relevant, currently the way the pick service is allowing to pick an object is if the object is withing 0.5 m in front of the robot or in 0.1m distance from the robot aside or behind. 
On the other hand, these Nav2 parameters regarding the goal_checker :
`xy_goal_tolerance=0.4` and `yaw_goal_tolerance=3.14` indicate that once the robot reaches the area of 0.4m from an object is going to stop and return success and the orientation of the robot in this reached pose can have maximum 180 degree tolerance.
This creates an inconsistency between the pick range and the acceptable goal tolerance. Assume the robot reaches the 0.4 distance of the object but the orientation it gets is in a way that the object is placed aside from the robot (not in front of it). Then the condition of distance < 0.1 will be checked in the pick server and if it is not less than that, the robot will fail to pick the object although it reaches the given goal pose as it was expected. 
To remove this inconsistency there are three possible ways:

1. reduce the `xy_goal_tolerance` to 0.1 to make sure regardless of orientation, the pick range and the Nav2 range are compatible with each other. I also suggest making the 0.1 slightly bigger like 0.25.
2. the students need to calculate the right orientation of the robot and then we can make sure that the object would be almost in front of the robot. Using this method, we need to reduce the  yaw_goal_tolerance to 0.5 since the exact orientation is going to be passed in a goal_pose. 
3. Remove the 0.1 aside and behind condition from the pick service. This means any objects within the pick range can be a candidate for pick and if the students do not provide the id of the object they want to pick for the server, the pick server chooses the first object provided in the list of the objects for pick and pick it. 
We need to reach on a consensus regarding this and based on the approach we select some minor changes may be needed. 
What do you think? @chcorbato @Rezenders @EGAlberts

Rezenders, 2025-03-24:
> the students need to calculate the right orientation of the robot and then we can make sure that the object would be almost in front of the robot. Using this method, we need to reduce the yaw_goal_tolerance to 0.5 since the exact orientation is going to be passed in a goal_pose.

In my view, this is the ideal solution, which would be solved by completing this issue.
I don't have strong opinions about this. If you all think it is better to go for simpler solutions like 1 or 2, it is also fine for me

EGAlberts, 2025-03-24:
I would like to throw into the mix that NavFn has a use_final_approach_orientation parameter (I recently discovered, it is off by default) which should help the students get their robot facing the object