kas-lab/krr_bringup #48: laser_filters dependency not declared

chcorbato, 2025-03-20:
> The workspace builds correctly, but then I get an error about `laser_filters`, I think a dependency not declared?
> help welcome @ArendJan @Rezenders 
> 
> ```
> ros2 launch krr_agent task1.launch.py 
> [INFO] [launch]: All log files can be found below /home/chcorbato/.ros/log/2025-03-20-15-06-00-046911-chcorbato-virtual-machine-28167
> [INFO] [launch]: Default logging verbosity is set to INFO
> [ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'laser_filters' not found
> ```
> 

 _Originally posted by @chcorbato in [#37](https://github.com/kas-lab/krr_bringup/issues/37#issuecomment-2740583327)_

chcorbato, 2025-03-20:
@ArendJan this is an issue in the `mirte-gazebo` repo, I dont know why I cannot transfer it there (maybe permissions)
I think I solved it in this pull request https://github.com/kas-lab/mirte-gazebo/pull/3

ArendJan, 2025-03-20:
https://filesender.surf.nl/?s=download&token=d2d0b01e-509d-49a3-ace5-e67f11b8075a also should work

chcorbato, 2025-03-20:
I did not, it is hard to keep up with the most updated procedure to test, we need somewhere the latest link, so I posted in a pinned post no our KRR project channel:
https://teams.microsoft.com/l/message/19:22c8cc9ad7384a00ae9e76a36c7612c8@thread.tacv2/1742484937952?tenantId=096e524d-6929-4030-8cd3-8ab42de0887b&groupId=f00cbfdb-47c0-4a8a-92ea-95952d6546a5&parentMessageId=1742484937952&teamName=RO47014%20Knowledge%20Representation%20and%20Symbolic%20Reasoning&channelName=Project%202024-25&createdTime=1742484937952

kvta1, 2025-03-20:
> https://filesender.surf.nl/?s=download&token=d2d0b01e-509d-49a3-ace5-e67f11b8075a also should work

I am also facing the `laser_filters` dependency issue when trying to run assignment 4 _KRR25_P4_PDDL_and_PlanSys_.

# Current Behavior

Tried this new singularity image and got the following:
```
kevin@kevin-LOQ-15AHP9:~/krr_ws$ singularity shell -p --nv ro47014_v3.0.sif 
Singularity:krr_ws> source ~/krr_ws/install/setup.bash
Singularity:krr_ws> source /usr/share/gazebo/setup.bash
Singularity:krr_ws> source /opt/ros/humble/setup.bash
Singularity:krr_ws> ros2 launch mirte_gazebo gazebo_mirte_master_navigation.launch.py 
[INFO] [launch]: All log files can be found below /home/kevin/.ros/log/2025-03-20-16-47-37-040105-kevin-LOQ-15AHP9-54
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'laser_filters' not found, searching:
```

Also could not manually install `ros-humble-laser-filters` as it seems that Singularity container is read only (?)
```
Singularity:krr_ws> apt update
Reading package lists... Done
E: List directory /var/lib/apt/lists/partial is missing. - Acquire (30: Read-only file system)

```
```
Singularity:krr_ws> apt list | grep ros-humble-laser

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ros-humble-laser-geometry/now 2.4.0-2jammy.20250121.124257 amd64 [installed,local]

```

# Environment
OS: Ubuntu 22.04
ROS2 Humble

ArendJan, 2025-03-20:
@kvta1 you can download the new singularity image from https://filesender.surf.nl/?s=download&token=d2d0b01e-509d-49a3-ace5-e67f11b8075a that one should have it installed

closed 2025-03-20 by Rezenders