kas-lab/krr_bringup #52: Deploy the template: copy it into all student groups on GitLab

chcorbato, 2025-03-21:
This cannot be done yet, we need to finalize the template first.
@Rezenders can you give @ArendJan the green light when ready?
It would be great if we can do it today.

I expect some improvement afterwards. Can we do it so that it is a fork of our repo, so that students can pull improvements from the remote?

ArendJan, 2025-03-21:
I can create the projects with all the history of github, but having github as en upstream remote is not possible

Rezenders, 2025-03-21:
Can't we just add a gitlab repo as another remote and push it to gitlab?

> This cannot be done yet, we need to finalize the template first.
@Rezenders can you give @ArendJan the green light when ready?
It would be great if we can do it today.

Well, if we wait until all the examples are ready it will take some time. 
What is strictly necessary to be done? We could already upload it and add some extra examples later.

ArendJan, 2025-03-21:
Yeah, but the 'upstream' remote won't be uploaded to gitlab afaik

chcorbato, 2025-03-21:
> Well, if we wait until all the examples are ready it will take some time.

Maybe more people need to be working on this. But why it will take some time? It is just moving a bunch of objects in a gazebo world and storing it, right?

> What is strictly necessary to be done? 

With the current 3 scenarios it is ok for the students to start working, as long as there are no blocking bugs.

> We could already upload it and add some extra examples later.

Yes, but then we need to make sure that students can update.

chcorbato, 2025-03-21:
@Rezenders  @ArendJan 

Shall we move the repo to gitlab, or fork and set up the automation @gavanderhoorn mentioned ?
At this point, I think we need the solution that can be implemented faster.

ArendJan, 2025-03-21:
putting it on gitlab and forking it there is the easiest probably. In any case it will be some manual steps, but not too much, probably half an hour in total for all the teams

chcorbato, 2025-03-22:
@Rezenders @ArendJan do we agree on moving the https://github.com/kas-lab/krr_agent repo to gitlab, then pushing to the students groups?
Can this be done on Monday?

When it is done, we need to tell the rest of the team.

ArendJan, 2025-03-24:
Will you do it @Rezenders or should I do it, its probably 2 mins work

chcorbato, 2025-03-24:
When done, we should notify everyone. Maybe with a project update

ArendJan, 2025-03-24:
Please send a teams message when I can move it?

chcorbato, 2025-03-24:
Green light sent
https://teams.microsoft.com/l/message/19:22c8cc9ad7384a00ae9e76a36c7612c8@thread.tacv2/1742823853927?tenantId=096e524d-6929-4030-8cd3-8ab42de0887b&parentMessageId=1742823853927&groupId=f00cbfdb-47c0-4a8a-92ea-95952d6546a5&owningTeamId=19:mm8Zes4Fa6s-gCzRPR6DfKtYFo-cixMdQELeoPHLwSw1@thread.tacv2

Rezenders, 2025-03-24:
Wait, let me remove the objects collision first

Rezenders, 2025-03-24:
Hmm, actually, disabling the collision of objects is not that simple as they will "fall" from the ground

chcorbato, 2025-03-24:
@gavanderhoorn can you create a main branch or give permission to Arend-Jan? (I’m limited by operating in my mobile phone)
https://teams.microsoft.com/l/message/19:22c8cc9ad7384a00ae9e76a36c7612c8@thread.tacv2/1742824159260?tenantId=096e524d-6929-4030-8cd3-8ab42de0887b&parentMessageId=1742823853927&groupId=f00cbfdb-47c0-4a8a-92ea-95952d6546a5&owningTeamId=19:mm8Zes4Fa6s-gCzRPR6DfKtYFo-cixMdQELeoPHLwSw1@thread.tacv2

chcorbato, 2025-03-24:
> Wait, let me remove the objects collision first

This seems involved and you are coordinating everything, someone else should look into it.

gavanderhoorn, 2025-03-24:
> can you create a main branch

done

chcorbato, 2025-03-24:
I have made the github repo read only, to avoid people to keep pushing their changes there. Not sure this is the way to proceed, suggestions are welcome.

ArendJan, 2025-03-24:
@chcorbato do you want me to create all the students repos? including the one that has something already?

chcorbato, 2025-03-25:
@Rezenders do we deploy krr_agent in the students' gitlab?

For the group with content, would that delete their existing content?

ArendJan, 2025-03-25:
@chcorbato no, it will only add it as an extra repo. No removal or overwriting of repos/code

chcorbato, 2025-03-26:
Agree with making krr_agent public.

@ArendJan cab you add the repo to all student groups?
@Rezenders can you coordinate with @buraksisman to write a short announcement about it on BS when done?

gavanderhoorn, 2025-03-26:
What's the benefit of making it public? IIUC, @ArendJan is going to create forks in the per-group subgroups. Students would already have the ability to see what happens in the source of that fork.

chcorbato, 2025-03-26:
Good point. Also, making it public is a commitment to maintaining it.
We should take that seriously, therefore maybe we should not do it till there is a feasible plan for that.

Rezenders, 2025-03-26:
> What's the benefit of making it public?

That we don't need to take any action if we update the repo with more examples, we can just tell them to look there.

gavanderhoorn, 2025-03-26:
Perhaps it's a misunderstanding on my side.

If *make public* means: give access to students, then that would make sense, and easily achieved -- and I assumed that would be what would be done already.

If *make public* actually means making the repository public (as in: from the current private setting), then I'm not sure that would be necessary to achieve what you describe.

Rezenders, 2025-03-26:
> Perhaps it's a misunderstanding on my side.
If make public means: give access to students, then that would make sense, and easily achieved -- and I assumed that would be what would be done already.

Makes sense. I agree with this, no need to make it public :)

ArendJan, 2025-03-26:
Indeed only make it readable for the students. 
I'm unable to fork it as im only a developer, not an owner, so cant create repos in their groups

Rezenders, 2025-03-27:
I added the `students-2425' group as a guest to the `krr_agent` repository. I set the expiration date to 30/04
I am closing this issue as it was completed

closed 2025-03-27 by Rezenders