kas-lab/krr_bringup #59: [BUG] Numpy version required for the pick action doesn't match the version in the singularity image

Rezenders, 2025-03-21:
```bash
[pick_place-12] Traceback (most recent call last):
[pick_place-12]   File "/home/gus/krr_ws/install/krr_mirte_skills/lib/krr_mirte_skills/pick_place", line 33, in <module>
[pick_place-12]     sys.exit(load_entry_point('krr-mirte-skills', 'console_scripts', 'pick_place')())
[pick_place-12]   File "/home/gus/krr_ws/install/krr_mirte_skills/lib/krr_mirte_skills/pick_place", line 25, in importlib_load_entry_point
[pick_place-12]     return next(matches).load()
[pick_place-12]   File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
[pick_place-12]     module = import_module(match.group('module'))
[pick_place-12]   File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
[pick_place-12]     return _bootstrap._gcd_import(name[level:], package, level)
[pick_place-12]   File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
[pick_place-12]   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
[pick_place-12]   File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
[pick_place-12]   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
[pick_place-12]   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
[pick_place-12]   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
[pick_place-12]   File "/home/gus/krr_ws/build/krr_mirte_skills/krr_mirte_skills/pick_place_node.py", line 22, in <module>
[pick_place-12]     from tf_transformations import euler_from_quaternion
[pick_place-12]   File "/opt/ros/humble/lib/python3.10/site-packages/tf_transformations/__init__.py", line 46, in <module>
[pick_place-12]     import transforms3d
[pick_place-12]   File "/usr/lib/python3/dist-packages/transforms3d/__init__.py", line 10, in <module>
[pick_place-12]     from . import quaternions
[pick_place-12]   File "/usr/lib/python3/dist-packages/transforms3d/quaternions.py", line 26, in <module>
[pick_place-12]     _MAX_FLOAT = np.maximum_sctype(np.float)
[pick_place-12]   File "/home/gus/.local/lib/python3.10/site-packages/numpy/__init__.py", line 324, in __getattr__
[pick_place-12]     raise AttributeError(__former_attrs__[attr])
[pick_place-12] AttributeError: module 'numpy' has no attribute 'float'.
[pick_place-12] `np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
[pick_place-12] The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
[pick_place-12]     https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations. Did you mean: 'cfloat'?

```

Rezenders, 2025-03-21:
For some reason, it works in the docker image built with the Dockerfile: https://github.com/kas-lab/krr_bringup/blob/main/Dockerfile

ArendJan, 2025-03-21:
My main recommendation is to update the package that uses outdated numpy stuff that is deprecated in 2021

ArendJan, 2025-03-21:
I'll try to update the old 0.3.1 package in the singularity script

ArendJan, 2025-03-21:
@Rezenders can you check with this version: https://filesender.surf.nl/?s=download&token=50b44bb6-27ee-406f-a5cf-b2c38aa6770b

Rezenders, 2025-03-21:
I tested it and it worked. Thanks @ArendJan

closed 2025-03-21 by Rezenders