kas-lab/krr_bringup #60: create evaluation scenarios for task 1

chcorbato, 2025-03-21:
First, we need to specify what are the changes in the world with respect to the 3 scenarios that already exist: https://github.com/kas-lab/krr_agent/tree/main/krr_gazebo/worlds

They need to be changes that make a hardcoded solution for the 3 scenarios that already exist not to work perfectly.
Can you work on this @fghzamani ?

We could give those 3 to the students.
We need to create the evaluation scenarios private to us, so they cannot be in the krr_agent repo, but on a repo private to us

Rezenders, 2025-03-21:
> We need to create the evaluation scenarios private to us, so they cannot be in the krr_agent repository but on a repo private to us

We can use the [krr_solutions](https://github.com/kas-lab/krr_solutions) repo for that

Today during the lecture, there were questions about what they can assume can change in  the evaluation scenarios:

1. Is the initial position of the robot always the same? If not, how can they know where it started?
2. Can the drop locations' position change or only the amount/which ones are included?
3. Are the drop locations always reachable?
4. Can the object types change?
5. Can they run over an object?
6. Objects in the door?

My view on this:
1. I suggest we don't change the robot's initial position. I don't see how this would contribute to the learning goals, and it would only add extra work for both sides (us and them) without any meaningful gain
2. I think it is fair to assume that the drop locations' position can change in each scenario. However, they remain the same during the execution
3. I suggest we make it always reachable
4. I suggest we keep the types we added
5. Disable collisions
6. 

Are there any other assumptions?

Rezenders, 2025-03-21:
6. Objects can be in the door. The robot needs to plan to pick it up and place it in another position before moving through the door. 
Implications: Change get_objects_in_room service to also return the objects between the doors. Change the service response to include `objects_in_room` and `objects_in_door`.

Rezenders, 2025-03-21:
@buraksisman can you write this assumptions in the overlead document?

chcorbato, 2025-03-21:
> * I suggest we keep the types we added

I want to add the option to introduce a new object in a task. I added this in the overleaf:

> All object types are known for a given task. Either they are from table \ref{tab:entities} or if a new type of object is included, information will be provided in the task message in the form of a tuple of strings, indicating the id of the object and the drop location where it normally belongs.

chcorbato, 2025-03-21:
Done, I had some ideas I wanted to include. @buraksisman can you double check and in general see if there are any inconsistencies/issue in the document?>

Rezenders, 2025-03-24:
> information will be provided in the task message in the form of a tuple of strings, indicating the id of the object and the drop location where it normally belongs.

From where exactly should this information be provided? I don't understand.
If we need to modify any services, we should create a new issue to describe what needs to be done

chcorbato, 2025-03-24:
See https://github.com/kas-lab/krr_agent/issues/2

alex-quin-gabriel, 2025-03-26:
@fghzamani @Rezenders according to my analysis, the objects that make the world blow up are bread and mouse.

chcorbato, 2025-03-31:
@fghzamani this issue is assigned to you. Can you please provide an update on the status asap?

fghzamani, 2025-03-31:
I pushed the evaluation scenarios to the GitHub repo Gustavo created: https://github.com/kas-lab/krr_evaluation
The test worlds for task1 and 2 are also available, and I will push them asap as I find the related Gitlab page.

chcorbato, 2025-03-31:
Close the issue when you consider it done, leaving a final comment where the commit that closes it can be found.

closed 2025-04-01 by fghzamani