kas-lab/krr_bringup #65: add wolrd with object in the doors

Rezenders, 2025-03-21:


Rezenders, 2025-03-21:
[image]

Done: https://github.com/kas-lab/krr_agent/commit/e3d9ca88d7ebb7fa6c0d413fc06efe30b1ecc2f1

chcorbato, 2025-03-21:
I do not understand this. The complexity of having objects at the doorways was intended to make Task 3 more interesting, not for tasks 1 and 2 (they already involve too much picking and placing)

Rezenders, 2025-03-24:
Hmmm, I understood you also wanted Tasks 1 and 2 to have objects between doors. If that is not what you want, I can remove this scenario from task 1 & 2 and only add it to the examples of task 3

chcorbato, 2025-03-24:
No need to remove it. But it will be an extra scenario, one of the evaluations giving  the highest score or extra points.

closed 2025-03-24 by Rezenders