kas-lab/krr_bringup #72: Replace hat, fidget spinner and hat for models that work for our problem

Rezenders, 2025-03-26:
Replace the hat, fidget spinner and mouse models for models that work for our problem
Check https://github.com/kas-lab/krr_mirte_skills/issues/9 and https://github.com/kas-lab/krr_bringup/issues/68

chcorbato, 2025-03-31:
What is the status of this @fghzamani ?

fghzamani, 2025-03-31:
Now the spinner is working. I fixed its problem. I also created a table of all the objects in the overleaf document (project instruction) but that table also includes the new objects in the evaluation worlds. We need to be careful with this table and comment the red columns before sharing the document with students.

chcorbato, 2025-03-31:
Have all objects in that table been tested @fghzamani @buraksisman .

If so, we can close this issue. Can you do that @fghzamani ?
If not, please indicate which objects need to be tested.

fghzamani, 2025-03-31:
I checked them all. They are working. But I think it is good if someone else also tests them. Maybe @buraksisman ?

Rezenders, 2025-03-31:
@fghzamani, I added the world files you sent me via Teams to the krr_agent repo. I cannot find a description of what should appear in the world, so I am sending screenshots here so you can check if it is correct

test_scenario1.1.world:

[image]

test_scenario1.2.world

[image]

test_scenario1.3.world

[image]

test_scenario2.1.world

[image]

Rezenders, 2025-03-31:
> I checked them all. They are working. But I think it is good if someone else also tests them. Maybe @buraksisman ?

I tested it. It seems to be working.

Rezenders, 2025-04-02:
Since this was merged, I am closing this issue

closed 2025-04-02 by Rezenders