kas-lab/krr_bringup #77: Navigation issue

buraksisman, 2025-03-31:
Robot navigates to desired location. However, when it is close to a furniture in the room, it stops. That is OK but there is distance to conduct the turning or go a little bit further. It might be adjusted.

Rezenders, 2025-03-31:
In addition, with the new nav2 params, the robot is stopping a bit far away from the object and with a wrong orientation (I know this was the intention), I would suggest returning to old params and let them figure out where to send the robot. It is very weird now.

chcorbato, 2025-04-01:
@burak will show to @fghzamani what is going on

closed 2025-04-01 by buraksisman

Rezenders, 2025-04-01:
This issue was closed. Does it mean we are not changing the navigation parameters anymore? 
Students reported a similar bug #83 where the robot gets stuck

chcorbato, 2025-04-02:
Please, @buraksisman, provide a comment that explains why you closed this issue.