kas-lab/krr_bringup #79: Freeze student groups on Gitlab so they cannot commit after April 11 23:59 CET

chcorbato, 2025-04-01:
Freeze student groups on Gitlab so they cannot commit after April 11 23:59 CET

is it possible to set this up @gavanderhoorn @ArendJan ?

ArendJan, 2025-04-01:
Yeah, it should be possible, either changing the users role to something else and/or making the main branch protected. Students are maintainers, so they can change those rules atm I think.

ArendJan, 2025-04-01:
You can change the timestamp of commits, so that's not 100% secure

ArendJan, 2025-04-01:
Then we should demote them to view-only or something

chcorbato, 2025-04-03:
Can we set it up so that students are automatically demoted to view-only April 11 23:59 CET (probably just April 12 if the parameter is just a date with no specific time)

ArendJan, 2025-04-03:
You can add an expiration date on the members page: https://gitlab.tudelft.nl/groups/cor/ro47014/students-2425/group_13/-/group_members they will then be removed as user. Then someone can add them back later the day after as view only.

chcorbato, 2025-04-03:
I know I should have specified this when the groups were created and students added @gavanderhoorn , I just did not think of this requirement